Northeastern-Electric-Racing/Cerberus #8: GPIO Expander Driver

nwdepatie, 2023-06-13:
This driver corresponds to the GPIO expander that monitors a lot of things on the PDU. This driver is gonna be a bit weird in that we probably want to be able to easily read each channel (ADC, toggle, etc) but then we also need to classify them on a higher level (i.e. turn motor on, read energy meter) so I'm not sure if that should happen on a driver level or monitoring level

nwdepatie, 2023-07-07:
Part number for this GPIO expander is PI4IOE5V9535ZDEX, probably should name stuff ```pi4ioe```

dyldonahue, 2023-07-15:
@nwdepatie im gonna create two layers, one that directly reads/writes and another that utlizes those to do specifc things like "turn motor on", read enery meter etc. If u or @HamzaIqbal69 wanna move the second layer to application later on feel free to, ill make it so that itll be portable but migth as well jsut do both here for n ow

closed 2023-09-17 by nwdepatie