Northeastern-Electric-Racing/Cerberus #43: Docker Image funky on M1 Mac

nwdepatie, 2023-09-22:
I was Lowkey anticipating this problem but have been unable to reproduce. Basically a docker image is usually just built for a certain architecture (i think) which makes it hard to deploy on ARM-based M1 macs when built in x86.

We had someone report that they were unable to start the docker container on M1 Mac and also was unable to build the Dockerfile locally. Reported a weird git error when pulling the latest version of ubuntu

Double check that an M1 Mac can pull the dockerfile and run it. Also verify that an M1 Mac can locally build it if needed

Sabramz, 2023-09-29:
@nwdepatie That person was me. I had to delete my docker config file, and the reason I think I had to do this was that my credentials were not configured right. The fix was deleting the docker config then logging in again. It built and ran after I did that.

nwdepatie, 2023-09-30:
Glad to hear, closing since this is resolved and I'll point people here if they run into a similar problem

closed 2023-09-30 by nwdepatie