Northeastern-Electric-Racing/Cerberus #169: [Controls] - Traction Control

nwdepatie, 2024-05-23:
### Current Features

In order for this to work, we need to have DTI callbacks working to get CAN data from it and integrate it into control algo.

We already have the torque calculation states enumerated in `torque.c` so it literally should just be a controls and verification thing

### Desired Additional Features

For traction control it can be a new algorithm or one that we got kinda working last year.

My implementation last year was to continuously check the "feedback torque" which was essentially just how much AC current we were actually drawing at that moment in time. TLDR, if the feedback torque didn't equal or commanded torque I would drop the commanded torque to equal the feedback torque and then slowly ramp up the commanded torque at a fixed rate.

To begin with, start with a simpler implementation, which imo the current draw is a good indication. Might want to look at our acceleration event's data to see what happens when we slip exactly

The nuance in this is "detecting the slip" and "how do we recover", which we might want to eventually fuse multiple different ideas for detecting slip together to get a good idea of what the car is doing (i.e. `slipConfidence = 0.2 * didWheelSpeedJustSpike + 0.5 * doesFeedbackTorqueEqualCommanded + 0.3 * isIMULinearAccelEqualToExpected` or something lol, not exactly but y'all get the idea of sensor fusion in a control pipeline).

nwdepatie, 2024-05-23:
You can use a PID controller to do the derating of torque based on the error between the two values maybe? https://github.com/Northeastern-Electric-Racing/Embedded-Base/blob/main/middleware/src/pid.c

nwdepatie, 2024-05-23:
Check this file for last years traction control (I called the function "controlLaunch") https://github.com/Northeastern-Electric-Racing/Embedded_Code/blob/main/MPU/src/pedals.cpp

Sabramz, 2024-06-23:
Note: After discussing with pTrain, this will only work well for acceleration.