Northeastern-Electric-Racing/Cerberus #194: Move functions in cerb_utils to embeded base

Sabramz, 2024-09-09:
### Description

These functions should be available to all C projects.

### Acceptance Criteria

- [ ] Functions in cerb_utils are in embedded base
- [ ] cerb_utils header and c file are gone from Cerberus
- [ ] The submodule is updated
- [ ] This is tested on the car

### Proposed Solution

Title

jr1221, 2024-12-04:
make sure they go in correct spots as some rtos dependent.

jr1221, 2025-02-28:
Make an rtos folder called `os`

jr1221, 2025-02-28:
1. create and cp 

os/src/rtos_utils.c
os/inc/rtos_utils.h

middlewares/src/debounce.c
...

2. add debounce.c and rtos_utils.h to cerberus Makefile

3. change names of includes as needed

4. build it with `ner build`

jr1221, 2025-02-28:
Base this off of `develop`

closed 2025-03-10 by maxr21