Northeastern-Electric-Racing/Cerberus #219: [Sense] - Create Accurate LV So Calcs

jr1221, 2024-11-05:
### Description

Create and send LV SoC calcs along with the voltage itself, in the same CAN message.

### Acceptance Criteria

SoC calcs are accurate to the discharge curve and relatively linear.

### Proposed Solution

Use the datasheet for the 24A LV battery, edit the CAN message spec in JSON.

harrison-e, 2024-11-06:
- Add SoC NetField to `read_lv_sense` CANMsg (`id 0x503`)
- Use [manufacturer Voltage to SoC function](https://www.aegisbattery.com/products/24v-20ah-li-ion-battery-pvc) to calculate SoC from `v_dec`
- Can be integer or decimal percentage 
- Sigmoid?

harrison-e, 2024-11-06:
- Normal charged voltage = 29.4V
- 18650 max charged voltage = 4.2V
- 29.4V/4.2V = 7 18650's
- Divide overall voltage by 7 to get avg voltage over all rows
- Model with sigmoid function to convert to SoC percentage

harrison-e, 2024-11-15:
Preliminary changes in `feature/soc` branch (equivalent changes in `Embedded-Base` on `feature/cerb_soc`)

PR #238 (2024-12-08): Added LV Sense SoC to CAN
## Changes

- Added LV Sense SoC calculation using logistic function to `Core/Src/monitor.c`
- Added `#include <math.h>` to `Core/Src/monitor.c`
- Changed CAN spec in `mpu.json` to include SOC NetField in `0x503` (See https://github.com/Northeastern-Electric-Racing/Embedded-Base/pull/212)

Closes #219
closed 2024-12-06 by harrison-e